Bastardkb/Charybdis #1: Are BTU adapters compatible with Charybdis 4x6 version?

JGrzybowski, 2021-10-11:
As stated in the title, are BTU adapter files in 3x5 directory compatible with 4x6 files?
If so, it might be convinient to put them in the separete direcotry for BTU mod.

LinuxBozo, 2022-01-15:
FWIW, I'm actively using the printable BTU mod on a 4x6 with great success. Works so much smoother than the roller bearings. Si3N4 ceramic bearing balls seem to work slightly smoother than stainless in the printable setup as well.
[image]

bstiq, 2022-01-15:
Hopefully this covers everything.
There's also a version of the cheaper btu mod I modified to use screws rather than press fitting.

Thanks for the feedback LinuxBozo!

Regards,

closed 2022-01-15 by bstiq

artlugovoy, 2024-06-13:
> FWIW, I'm actively using the printable BTU mod on a 4x6 with great success. Works so much smoother than the roller bearings. Si3N4 ceramic bearing balls seem to work slightly smoother than stainless in the printable setup as well. [image]

Tnis is awesome! Please could you share the adapter STL?