Bastardkb/Charybdis #30: Keyboard 4x6

brunooomelo, 2023-04-02:
do you have any 4x6 model keyboard without balltrack?

Boshimanu, 2023-04-03:
Hi, I'm searching for the same thing, but I do not find it... Are we missing something? 
I don want to buy a kit, but I wanted to print it first to see how it works for me.... Awesome Project. Thanks!

bstiq, 2023-04-03:
Hello 

Yes, check out the Scylla:
https://github.com/bastardkb/scylla

Regards,

hughrawlinson, 2023-06-13:
Hey, to follow up on this repo, the 4x6 model here seems to only contain the left hand model with the trackball as a step file, the right hand model with the trackball as an stl file, and supporting files like the trackball adapter and the tent. To be a bit clearer with your previous comment - on [the website](https://bastardkb.com/product/charybdis-kit/), which shows a picture of a Charybdis set with the trackball on the right and a non-trackball left hand - is that left hand using the Scylla model?

Thanks for all your work on this!

bstiq, 2023-06-14:
Hey, 

If you want the right hand on the trackball, you should mirror the charybdis file.
You can do that using eg. prusaslicer.

For the left hand, you should use the scylla model yes.

All files are reversible, including plates and tents. 
For tents, the charybdis ones are compatible with the Scylla.

shayded-exe, 2023-12-07:
> Hey,
>
> If you want the right hand on the trackball, you should mirror the charybdis file. You can do that using eg. prusaslicer.
>
> For the left hand, you should use the scylla model yes.
>
> All files [...]

Could you please explicitly mention this somewhere in the readme? I didn't realize until after printing that I printed the left handed trackball! The files also should be named left/right for clarity.

bstiq, 2023-12-11:
> Could you please explicitly mention this somewhere in the readme? I didn't realize until after printing that I printed the left handed trackball! The files also should be named left/right for clarity.

I'm not sure what you mean here, the Charybdis STL file is right handed already.

bstiq, 2023-12-12:
The STEP file is not meant for printing, but for editing.
I understand the STL file not oriented correctly might be an issue for printing as it sometimes is used for beginners, so when releasing a new charybdis version I take the time to mirror it properly.

If you're going to use the STEP file it probably means you're modifying the keyboard, so I'm assuming you know what you're doing.
I'm providing it out of convenience for advanced tinkerers.

shayded-exe, 2023-12-13:
> The STEP file is not meant for printing, but for editing. I understand the STL file not oriented correctly might be an issue for printing as it sometimes is used for beginners, so when releasing a [...]

Do consider that most slicers support STEP now and in doing so can produce better optimized gcode (mainly for circles and arcs). I always print the STEP file when one is available.

Not a big deal either way :)