Bastardkb/Charybdis #44: Request for Kailh Hot-swappable PCB Socket  boards and USB C board to board connecters

starbuxed, 2024-03-13:
Kailh Hot-swappable PCB Socket.... as they are easier to solder.  The mil sockets dont work great with a hot plate  and the kaili sockets are easier to solder with shaky hands. I have nightmares still from iris v4 and the mil sockets. 

Also  update the board to board connectors to USB C  as its much easier to find Usb C cables than trrs. especially with small patch cables. 

It would bring the project in line with most current keyboards.

bstiq, 2024-03-13:
Hey,

We won't support hotswap - you can read more about it here:
https://docs.bastardkb.com/help/faq.html#hotswap
This is also why none of the other 3d keyboards won't support it (well, those manufactured in more than very small quantities).
If you want to fork the PCB project and modify the files, we would gladly accept a PR to link your repo as a mod.

For the board-to-board connection, we are sticking to audio cable because:
- it's a standard amongst the split keyboard world
- there are a lot of fancy audio cables, for aesthetic purposes
- having usb-c to usb-c brings *a lot* of problems (1)

(1) mostly design issues, to make sure plugging the wrong cable in the wrong port won't fry something. Plus all the confused users...

Hope it makes sense!

closed 2024-03-13 by bstiq

starbuxed, 2024-03-13:
> Hey,
>
> We won't support hotswap - you can read more about it here: https://docs.bastardkb.com/help/faq.html#hotswap This is also why none of the other 3d keyboards won't support it (well, those [...]

Well  a majority of the spilt keeb is moving to usb-c or wireless....like  Beekeeb, keebIO,.... Now we are talking about people building keyboards.  You really think people are going to plug them in wrong.  And then its just a matter of wiring the power correctly.  Which is why you still see trrs jacks on splits trrs is simple. and people are unwilling to wade into the complexity that is usb c.  also there are a ton of established projects using the trrs... so they copy/ use that too. as far as the custom cables there are plenty of people who make the c to c ports as well... So as long as you wire them up correctly there is no risk.

sevenrats, 2025-01-12:
url is now https://docs.bastardkb.com/faq.html

sevenrats, 2025-01-12:
https://ryanis.cool/cosmos/plum-twist/