Bastardkb/Charybdis #54: broke documentation when you silently migrated to 3mf without warning

sevenrats, 2024-12-13:
commit 2dad0ca removes the file linked in the docs and replaces the stl with a 3mf file. 3mf is dogshit on linux right now and it would be *awesome* if you could continue providing stl files for people who don't want to rip all their hair out.

bstiq, 2024-12-13:
Hey,

- Can you link where the docs broke? I don't expect a 3mf vs stl difference is an issue, it's difficult to track changes like that
- Can you expand on the lack of compatibility of 3mf on linux? As long as you have a slicer (eg. Prusaslicer, Orcaslicer..) it should be fine

sevenrats, 2024-12-13:
the linux problems were just my ignorance,  I was down one of these rabbit holes:
https://github.com/SoftFever/OrcaSlicer/issues/6800
https://github.com/3MFConsortium/gladius/issues/3

but I understand now how the migration was the correct decision.

https://github.com/Bastardkb/Charybdis/pull/55

bstiq, 2024-12-14:
We can eventually add a note about the STL/3MF file interchangeability

bstiq, 2024-12-29:
Fixed with https://github.com/Bastardkb/Charybdis/commit/0fbf4346a73e6e77c5d61bf478446d3cc5d47789

closed 2024-12-29 by bstiq