Bastardkb/Charybdis #56: Change trackball polling rate?

Rouji, 2024-12-19:
Hi!
I recently got a 120hz display and noticed the trackball in my Charybdis seems to only update at 60-ish Hz. Not sure how I can actually measure that, but it's definitely less than my other mice. 
I tried changing `USB_POLLING_INTERVAL_MS`, but that didn't do anything. 
Anything I can do to make it poll faster?

Rouji, 2024-12-30:
Aha!
Figured out it's not a setting, but something to do with RGB. If I turn RGB off, my pointer is nice and smooth. Turn it on, and it goes down to the horrible inconsistent 50~60Hz.

Guessing the controller can't keep up with both things at once? :thinking:

Rouji, 2024-12-30:
The lagginess happens when:
- RGB is on
- AND the auto mouse layer is active
-  AND CHARYBDIS_CONFIG_SYNC is on

...which in my case was always the case whenever I used the trackball. I have no idea why this combination of things makes it lag, but disabling CHARYBDIS_CONFIG_SYNC seems to completely fix this

Rouji, 2024-12-30:
actually, it might be on any layer that's not the base layer. not specifically the auto mouse layer

bstiq, 2025-01-06:
Hello,

I don't have a high refresh rate display on hand, so I will ask around in the discord community.

bstiq, 2025-01-06:
To help further, can you detail:
- your hardware (RP2040 splinktegrated? blackpill? etc)
- if you are using the latest firmware code from the BKB QMK fork (on bkb-master branch)
- if you tried using a different computer / screen already

Even on old firmware the default polling rate should be at least 125Hz, so we are investigating. 
It could be an issue with half duplex communication.

Could you maybe link us your uf2 image here so we can test on our end?

bstiq, 2025-01-06:
According to KarlK90:

> Is the USB connection on the same side as the Trackball? Otherwise all the mouse reports are synced over the serial connection which might lead to congestion. If that solves the issue the problem is probably the limited bandwidth. Bumping the serial speed would help with that.

daskygit, 2025-01-06:
Does the issue also appear if you lower the brightness? Turn it pretty much all the way down.

Rouji, 2025-01-06:
Ah yeah, kinda forgot there's different versions of things. 
This is:
- splinky v3
- just basic upstream qmk
- usb on same side as trackball

and it happens on different PCs, and on all brightnesses (except off)

bstiq, 2025-01-07:
Some tests by Dasky:
[image]

> while not brilliantly consistent, it's certainly smooth to my eye on a 240hz monitor

Did you also try using different cables? Maybe a different screen if you can as well.

Rouji, 2025-01-07:
Yeah I tried different cables, computers, screens, and OSs. 

I'm not really dependent on CHARYBDIS_CONFIG_SYNC, and disabling that completely fixes this problem, so I'm pretty much happy again. 

But I'll try the default firmware and stuff just as a matter of interest

Rouji, 2025-01-07:
Can't reproduce with the vendor image, but I'm not sure the circumstances even apply with that(?)

bstiq, 2025-01-09:
So this seems like it's an issue with the config sync.
As it's an isolated bug, I'll close this for now, feel free to reopen if there's further issues or someone else with the same issue.

closed 2025-01-09 by bstiq